sedmenti/Orbion_Space_Mouse_OLD #7: Possible enhancement to include lateral input

Stargrove1, 2022-01-18:
Not an issue, just an idea I'd like to ask about.

Arduino has two free AIs, this would allow to add a second joystick input. If the second joystick gets mounted upside down on top of the existing one, it might be possible to create a lateral input by adding both joysticks absolute value for lateral movement, while adding the values for existing input for both axis. The joysticks need to be connected by a kind of tube, maybe printed and glued.
See below sketch
  
[image]

FaqT0tum, 2022-01-19:
I like the idea, but in this way you would raise the knob too much, in addition to the fact that the movement will always be on a fulcrum and never linear

However, I am already working on a version with 6 axes managed by optical sensors which are the best choice for precision and reliability.

Stargrove1, 2022-01-19:
You are right, it would require to lower it down in the base. The loss in height during lateral movement would be not be very big, only ~3mm. 

But if you are already working on a different solution, that's even better. What kind of optical sensor will be used?

FaqT0tum, 2022-01-19:
3mm can be many.  for the optical sensor for now I can tell you a little, I am still looking for an optimal solution for the simplified realization and functionality

Stargrove1, 2022-01-19:
I hope to see the upgrade soon. Thank you for your effort

closed 2022-01-19 by Stargrove1