sedmenti/Orbion_Space_Mouse_OLD #8: Suggestion

ssombra, 2022-01-18:
Is it possible to add the possibility to chose the Mouse orientation? As it is now (at least in my build), the mouse should be used with the screen at the left. It would be very useful to have the option to change the orientation so the user can choose how to grab the mouse, with the screen at the front for instance. I actually changed the code to do that, but an option in the menu could be a good solution.

FaqT0tum, 2022-01-19:
you could, but once you have set the direction in the sketch you do not
change it anymore during its use, so I think it is of little use

closed 2022-01-19 by FaqT0tum

ssombra, 2022-01-19:
> you could, but once you have set the direction in the sketch you do not change it anymore during its use, so I think it is of little use

Although you are right, there could be some situations where it could be convenient to change it, for instance using a different USB port (or computer) and having the cable coming from other side. 
Anyway, thanks for your answer