sedmenti/Orbion_Space_Mouse_OLD #9: Knob axis base mount (the thing you put on the joystick-shaft)

binxyde, 2022-01-19:
Hi there! 

You did an awesome job with this! I love your precise design an especially the fact that you only used modules that are available quick and easy! :)

The last days I rebuilt this cool space mouse, and I would like to suggest a little mod on the knob axis part.
I made the mount for the joystick-shaft 3mm longer and made a little bit more space for it inside. 0.2mm in every direction.
Then you can push it easily onto the joystick and you have a little bit space to calibrate the alignment to the base.
A drop of hot-glue will hold it tight.

With the original design I destroyed one of my joystick modules while trying to push it onto the shaft :/

Best wishes,
binxyde

[image]
[image]

stevedee78, 2022-01-21:
Can you share the STL from your Knob_axis, please?

FaqT0tum, 2022-01-21:
the holes are very dependent on the print quality, if the printer creates small blobs, internal seams or is not properly adjusted, the holes may be wider or narrower ... I assume this is your case as it is the only signal that has arrived.  all files have already been released in the STL section, including the REV.  2

closed 2022-01-21 by FaqT0tum

binxyde, 2022-01-21:
@FaqT0tum yeah...the printer (creality cr-10s pro) was calibrated just before that print...but i saw that i have forgotten to use your extrusion width parameters. maybe thats the fault.

@stevedee78 i can't attach stl files here. But FaqT0tum published the fusion file. so its an easy thing to mod it by yourself.