sedmenti/Orbion_Space_Mouse_OLD #12: need for 2 version of the joystick too: different mounting holes, different joystick rod

netweaver1970, 2022-01-21:
Hi,
small question, just this morning I made a modification to the v1.1 knob axis for a different (same name as far as I know) joystick. This was the joystick I had in my arduino stuff drawer. It has a thin 2x1 mm metal rod on top instead of the thick plastic stub you made the design for. I used the V1.1 Fusion360 file.
And just then I saw you released 2.0, with 2 variants for different encoders :)

I could add another variant for my (and maybe other's) joystick. I could do it using the STEP file but a bit cleaner would be to use the Fusion 360 file. If you don't want to release the Fusion file yet on Github, maybe you can share it using Fusion360 directly, I'll only add the new variant of the knob_axis and the corresponding mounting hole changes. All images below are from the STEP file.

this is the situation with the 2 joysticks superimposed, you can see the holes don't align.
[image]

And this is the top part, the rods are totally different. My joystick is active:
[image]

If you would like to do it by yourself, it's also fine, this is the source link for the 3d model: https://grabcad.com/library/dual-axis-xy-joystick-module-with-push-button-1

Thanks for the good work.

Cheers.

FaqT0tum, 2022-01-22:
Hello, everything has been released, unfortunately I forgot it due to sleep. Unfortunately in this case I will not release any variant because this joystick is not good for mechanical reasons
However, I appreciate your availability. thank you very much

closed 2022-01-22 by FaqT0tum

drakengard780, 2022-01-30:
hi i have the same joystick i modified the project for the 14 mm. thise is the [https://www.thingiverse.com/thing:5224533](url)