sedmenti/Orbion_Space_Mouse_OLD #16: Not Recognised

Aesavyx, 2022-01-24:
After flashing the firmware, the pc callths the device "Unknown USB Device" in device manager and it no longer shows on the COM ports. The screen works but it doesn't do anything in any programs.

Clay2F, 2022-01-25:
What program did you use to create the sketch?
I created mine with VScode and PlatformIO and had to add the libraries manually or use others.

Aesavyx, 2022-01-25:
I used arduino IDE.

FaqT0tum, 2022-01-26:
> Ho usato arduino IDE.

the problem has been solved?

Aesavyx, 2022-01-26:
I think I just had a faulty pro micro

closed 2022-01-26 by Aesavyx

drakengard780, 2022-01-27:
Same problem . After flash code.not recognize . I can t reflash

edifabio-72, 2022-01-28:
Something similar happened to me too. I solved it by disconnecting and reconnecting the usb cable. The arduino board should change usb port. It should now reflash correctly.

drakengard780, 2022-01-28:
When i write code , After thise the USB no recognized . So not work . What can i do ( sto perdendo la capa 😂)