sedmenti/Orbion_Space_Mouse_OLD #18: Rosters encoder needs more clearance

spuder, 2022-01-29:
The rotary encoder I received has solder that slightly interferes with the print

It would be good to add a little more clearance for resistors and solder joints

[image]

spuder, 2022-01-29:
I printed the 14 mm version from version 2.1 and the holes are also not aligned.

[image]

[image]

Clay2F, 2022-01-29:
turn the printed part.
the hooks at the sides must show to the bottom

Clay2F, 2022-01-29:
[image]

closed 2022-01-29 by spuder

spuder, 2022-01-29:
[image]

Thanks for the info. I've flipped the part and I'm finding that the holes still don't line up. I've drilled them out and trimmed excess material but it still isn't quite lined up.