sedmenti/Orbion_Space_Mouse_OLD #19: LCD holes are 27mm

spuder, 2022-01-29:
The lcd I received has holes that are 27mm apart, but the pegs on the front plate are closer to 29mm

[image]

[image]

spuder, 2022-01-30:
[image]

FaqT0tum, 2022-01-30:
the components in question are of low quality so even if the links are correct it may happen to receive parts with different tolerances!  therefore you can very well understand that I cannot create different components for each tolerance that the Chinese are wrong in production… there are many people who testify that the project quotas are correct, except in these cases like yours.  so please do not create a thousand issues Tricket for all your difficulties regarding the components ... Forgive me but unfortunately I am powerless in these cases

closed 2022-01-30 by FaqT0tum

spuder, 2022-01-30:
I'm sorry if I offended you by reporting a problem. I did not intend to open "thousands of issues". I opened 2. 

I'm working on solutions to these 2 problems so that other people do not have the same trouble. I apologize if you thought I was asking _you_ to fix it. That is not my intention. 

How do you want me to contribute back these modified versions with fixes? Shall I upload STL fixes, or fusion 360 files?

FaqT0tum, 2022-01-30:
I'm sorry if I made you understand this, I was not offended 😋 ... mine is just a way of managing ... I like it when people contribute in order to improve the project ... if you have a real and working solution you can make a pull request or another well documented issues ticket so that I can add your improvements to the project

VinnyCordeiro, 2022-02-17:
I was about to open an issue about that because I also had alignment problems. The thing is: I bought all the components using the links provided by the author on the README.md of the version 1.1, it was to be expected that the printed parts would fit those components, as they were used as the reference. And I know that the body is already on version 2.2, but looking at the 1.1 version CAD file I can see some concerning alignment issues, like the ones on the screen, or the alignment of the holes on the encoder with the ones on the encoder holder, which are plainly wrong.

Granted, the newer version seems to use a different way to connect the encoder, but the screen is still off: the horizontal distance between the screw holes on the display is about 29 mm, while on 2.2 CAD it's 30.35 mm; and measure vertical distance is about 27 mm, while on CAD we have 28.6 mm.

I'd do these changes, but since the chosen license does not allow redistribution, I'm reporting the problem so that the author can fix it. I could ignore the license and solve all that, but I won't.

FaqT0tum, 2022-02-17:
hello, I'm sorry for the inconvenience but unfortunately for reasons of shortage of electronic components, the Chinese are spending a bit at random what is left in their warehouses.  other people have also had this problem and I want to solve it, for problems of necessity I haven't done it yet but I should have finished with the most important improvements of the firmware and encoder area.  as you can see I always try to be active in the resolutions ... I ask you if you please, can you send me the exact measurements between the holes of your oled so that I can also compare them with other users ... for other problems or doubts you can access the discord which will be easier for them speak