sedmenti/Orbion_Space_Mouse_OLD #20: Knob is too short

spuder, 2022-01-30:
I printed the 2.1 knob (short) and plate, but I'm finding the following problems

1. The pieces do not align
2. The Encoder is too short (I bought the exact one linked in the readme.

Did I print the wrong part? Or am I missing a piece?

[image]

[image]

[image]

[image]

spuder, 2022-01-30:
The knob is assumed to be 27.6 mm tall, but the part referenced in the readme are only 23mm tall

[image]

FaqT0tum, 2022-01-30:
the components in question are of low quality so even if the links are correct it may happen to receive parts with different tolerances!  therefore you can very well understand that I cannot create different components for each tolerance that the Chinese are wrong in production… there are many people who testify that the project quotas are correct, except in these cases like yours.  so please do not create a thousand issues Tricket for all your difficulties regarding the components ... Forgive me but unfortunately I am powerless in these cases

closed 2022-01-30 by FaqT0tum

jerzytuleja, 2022-02-04:
The problem is that Chinese producers offer KY-040 with long (with and without threads) and short shafts, with the same name of the product. 
I needed to buy 3 sets to get a proper one. I think the solution proposed by @spuder will solve the problem.

AFUDIrk, 2022-02-04:
The encoder with the long thread is better.  This can also be fastened with
the nut.  Thus, the soldering points are not so heavily loaded.  Here's my
version.  https://github.com/AFUDIrk/Orbion_Space_Mouse-2.2

Mit freundlichen Grüßen
Dirk König

FaqT0tum, 2022-02-04:
> The problem is that Chinese producers offer KY-040 with long (with and without threads) and short shafts, with the same name of the product. I needed to buy 3 sets to get a proper one. I think the solution proposed by @spuder will solve the problem.

I am looking for a simple and universal solution