sedmenti/Orbion_Space_Mouse_OLD #24: Fusion 360 Pan & Orbit

helicopterguru, 2022-02-02:
Good Afternoon.

Thank you for this great project, I am having some fun messing around with this and will soon be 3D printing the parts to assemble it.

Currently I am simply running the electronics on the bench and all is working ok, I to am seeing the rotary encode issue. But from another product I have with a similar issue, I find rotating the pot slower helps, not sure if this can help you smooth it out or not.
But my problem, or lack of understanding, is that within Fusion360 the joystick changes between pan and orbit without my input. So I am curious if I am doing something wrong, or there is a way to lock the joystick to control one or the other function?

Cheers.

Ian Contessa

FaqT0tum, 2022-02-02:
unfortunately, the encoder in question is very limited in the quality of the output, I did everything possible to make it as smooth as possible but there is a hardware limit that I cannot overcome with the code alone.  to switch from pan to orbit I have set the central key (encoder) which, if held down, makes me access the pan while moving the joystick, upon release I return to orbit mode, you can set this in the mouse settings that you access with the central button of joystick which once assembled is the second hardest click of the knob

Matt (Faq_t0tum)

helicopterguru, 2022-02-02:
Hey Matt.

Thank you for your time replying.

Yeah I can understand that, the product my electronics engineer designed for me using the rotary encoder also suffers similar issues.
Ah ha great, I will go an check now with holding the encoder pressed, I was trying a few things to figure out what was going on, but never thought about that. Ideally I now need to print the casing for it to make it neat.

Thanks for the fun projects.

Ian Contessa

FaqT0tum, 2022-02-02:
Imagine it's a pleasure ... this you find on the GIT home page will help you navigate the functions 

[image]

helicopterguru, 2022-02-03:
Thanks Matt.

Well apart from the SH1106 screen which fills the space better and the 8mm button, both items on order, I have a working Orbion_Space_Mouse :-)

Two requests if I maye;

1. Within the INO file can I swap round the pan and orbit, so by deffault it orbits and then pressing the gimbal button changes to pan?
2. Would you be able to upload the STEP files of the Knob_Axis_17mm and the Screen Cover please?

The Knob_Axis I want to make about 1mm thicker at the most so I can recess the area below the potentiometer a little more as I had to cut away from material to miss the smt resistors on the back of the PCB. 
The screen cover, currently I am using an SSD1306 display so I can modify it to fit that OLED and also replace the 3D printed locating pins with small screws, as mine broke away easily.

Thanks.

Ian Contessa
[image]
[image]

macgyverek, 2022-02-03:
Hi, I rewrite all your code and have new one with better encoder library. I want send you my code to you, but I can't found your email address. 
[image]

macgyverek, 2022-02-03:
What I made:
1. Rewrite all code
2. Added Encoder Library
3. Added Button Library
4. Rewrite menu
5. Rewrite memory
6. Added LED backlight
7. On and auto off function
8. Pan or Orbit mode switch
9. Code optimalization

Kretdiop, 2022-02-03:
> What I made:
> 
> 1. Rewrite all code
> 2. Added Encoder Library
> 3. Added Button Library
> 4. Rewrite menu
> 5. Rewrite memory
> 6. Added LED backlight
> 7. On and auto off function
> 8. Pan or Orbit mode switch
> 9. Code optimalization

May i please code ? :) and some details about led ?

macgyverek, 2022-02-03:
https://drive.google.com/drive/folders/1A9cxjMF87qoBKpCRosQySsRSoncSymva?usp=sharing

I used 2x 3mm Blue LED connected in series + 10 Ohm Resistor. Kathode is connected to GND, Anode to PIN18. LCD Power is connected to PIN4. Now I can control LCD power directly from ATmega. The current is sufficient.

FaqT0tum, 2022-02-03:
> What I made:
> 
> 1. Rewrite all code
> 2. Added Encoder Library
> 3. Added Button Library
> 4. Rewrite menu
> 5. Rewrite memory
> 6. Added LED backlight
> 7. On and auto off function
> 8. Pan or Orbit mode switch
> 9. Code optimalization

Very interesting! as soon as I can, I hope tomorrow, I'll take a look at it, but for the LEDs I already intended to insert them, but some neopixels so that the color can be managed from the menu

macgyverek, 2022-02-03:
I changed Encoder mounting and LCD cover. 
[image]

FaqT0tum, 2022-02-04:
> 

why did you create that fixing bayonet? the knob should be kept as light as possible so as not to worsen the quality of use.

FaqT0tum, 2022-02-04:
I don't like visible screws

FaqT0tum, 2022-02-04:
I close for lack of relevant problems

macgyverek, 2022-02-04:
I didn't change position of knob I only extrude this part. Why?? Encoder play left/right between legs and encoder hausing. So encoder axis move left/right too.  When knob was rotate and gaps between base and knob was minimal, parts was rubbing.

FaqT0tum, 2022-02-04:
understood, in the mounted ones I have not encountered this oscillation problem, I will pay more attention next time so as to create a solution

helicopterguru, 2022-02-04:
Hey Guys.

I've had the knob rubbing and can see it's slightly offline on the shaft of the rotary encoder, but feel this is because of the resistors on the back of it's PCB not having enough clearance in the Knob Axis piece to sit flush on top. Thus the pot isn't exactly vertical which makes the knob rub in certain positions around the 360 degree rotation. 
I've redrawn the Knob Axis, making it 0.8mm higher and this allows me to make the recesses for the underside of the pot deeper, so it should sit flat on top which will make it perpendicular to the plate.

If we can have the STEP files to play around that will be great, as working with converted STL's is a PITA.

I'm enjoying my build, looking forward to any additions.

Ian Contessa

FaqT0tum, 2022-02-04:
> Hey Guys.
>
>
>
> I've had the knob rubbing and can see it's slightly offline on the shaft of the rotary encoder, but feel this is because of the resistors on the back of it's PCB not having enough [...]

cad files have always been available in both STEP and F3D

helicopterguru, 2022-02-04:
> cad files have always been available in both STEP and F3D

DOH! Sorry, I had a senior moment and didn't look in the CAD folder.

Have a good weekend.

Ian Contessa

FaqT0tum, 2022-02-05:
> I changed Encoder mounting and LCD cover. [image]

[image]

The next release will have this modification which will make the encoder more stable and solid

helicopterguru, 2022-02-05:
Hey Buddy.

I've made it even more stable again and changed the knob axis to match. Want these parts?

Ian Contessa
[image]

helicopterguru, 2022-02-05:
[image]
[image]

helicopterguru, 2022-02-05:
I'm wondering if we could make a small round PCB design that bolts to the knob axis plate, then the pot, resistors and some LED's solder to it, with a JST or other connector to bring the wires up to the underside of the knob plate. Then the rotary part is a seperate module and we can easily fit LED's up here.

Ian

helicopterguru, 2022-02-05:
[image]

blaghislain, 2022-02-05:
@helicopterguru Very intesresting.  Is it possible for you to github fork this project in order to share all your change and CAD remix?

FaqT0tum, 2022-02-05:
> [image]

[image]

I'm just fixing this area up a bit. I'm adding Neopixel which will also be implemented in the menu. 3 m3 screws are enough and advance, 4 are useless, there is not much force to counter, the encoder will break first.
I believe that the final design will be a little different from this one, I want to reduce the parts for the weight and overall dimensions factor as well as making everything more universal as possible that the continuous unknowns that the Chinese create in production

helicopterguru, 2022-02-05:
Raised middle ring in upper body piece to try and disguise joystick as much as we can
[image]

helicopterguru, 2022-02-05:
> [image]
>
> I'm just fixing this area up a bit. I'm adding Neopixel which will also be implemented in the menu. 3 m3 screws are enough and advance, 4 are useless, there is not much force to counter, [...]

Hey.

Nice, the neopixel will be fun.

With the fixing, 3 will be good if you can move the 3rd one on the opposite side of the two existing holes, so it clamps in a 3 pointed star pattern.
I'm thinking to rotate the pot 180 degree's and the wires soldered to it go straight out the bottom of the knob axis plate so they don't go up into and over to reach the front. Going further, a JST-XH connector would solder to the pots wire points, it's a little big though, but would then create a connection point underneath for easy maintenance.
As well as the 4 screws I also reduced the PCB recess in your existing ENC Holder so it did actually clamp the PCB to the surface.

Yeah consistancy in the parts available can be a problem, I had one potentiometer which I realised today had mouting holes 3mm closer to each other than the one you linked to and I purchased. So that shows how somethings can be different.

Ian Contessa

FaqT0tum, 2022-02-06:
> Raised middle ring in upper body piece to try and disguise joystick as much as we can
> 
> [image]
> 
> 

what is the use if you have to use Foam?  then did you see the bellows made by HiNATA that I posted on the project?

FaqT0tum, 2022-02-06:
I did not use a star pattern because there are hooks to hold the cables in place, and in any case nothing changes, I repeat that there are no forces to counter so you need to do a careful job on the positioning of the components, as long as everything is centered and no weird vertical angles

helicopterguru, 2022-02-06:
> I did not use a star pattern because there are hooks to hold the cables in place, and in any case nothing changes, I repeat that there are no forces to counter so you need to do a careful job on the positioning of the components, as long as everything is centered and no weird vertical angles

Afternoon.

Yeah you have to avoid the cables going to the pot, so this limits you to some degree. So far I'm liking the 4 point fixing, over-engineered maybe, but it clamps the pot nice. 
Currently reprinting the base and supp_ele peice, as I noticed the spindle of the joystick is 0.5mm out of alignment in the X & Y axis on the 3D printed part. Small difference, but I noticed it on the actual peice and the calipers show the slight difference to the inside of the upper body part. So have changed the supp_ele and the brass inserts.
I've also opened up the sides of the wiring guide channel so that I can have some wires coming out the sides under the supp_ele peice to make wiring easier.

Ian Contessa
[image]

helicopterguru, 2022-02-06:
A future try might be a different Knob Axis with the pot rotated 180 degrees and soldered wires going straight out the bottom down into the main body. I'm still liking the idea of a small connector here though, so not rushing to print this one out yet.
[image]

helicopterguru, 2022-02-06:
Meant to say, no foam yet under the knob, I will get some soft foam and laser cut a ring to fit under it. A bellow would be nice, I will have to go looking. At least with the raised middle ring, this will stop the foam moving around too much if I cut the inner diameter the same size.

FaqT0tum, 2022-02-06:
> 

actually making a side buttonhole is a good idea

FaqT0tum, 2022-02-06:
> Un tentativo futuro potrebbe essere un diverso Knob Axis con la pentola ruotata di 180 gradi e i fili saldati che escono dal fondo verso il basso nel corpo principale. Mi piace ancora l'idea di un piccolo connettore qui, quindi non correre ancora a stamparlo. [image]

this is not good, you stress the soldered cables too many by breaking them

FaqT0tum, 2022-02-06:
> Intendevo dire che non c'era ancora schiuma sotto la manopola, prenderò della schiuma morbida e taglierò al laser un anello per adattarlo sotto di essa. Un muggito sarebbe carino, dovrò andare a cercare. Almeno con l'anello centrale rialzato, questo impedirà alla schiuma di muoversi troppo se taglio il diametro interno della stessa dimensione.

[image]

closed 2022-02-06 by FaqT0tum

helicopterguru, 2022-02-06:
Oooh is that a 3D printed TPU bellow/rubber boot?

FaqT0tum, 2022-02-06:
> Oooh è uno stivale a soffietto/gomma in TPU stampato in 3D?

yep

FaqT0tum, 2022-02-06:
it also improves usability, a Japanese user called HiNATA did it

helicopterguru, 2022-02-06:
Very nice, I will have to go searching for it. I can image it does smooth the action up a bit, as there is some additional resistance, but not enough to stop the movement.
Super fun little project this thanks.

Ian Contessa

AFUDIrk, 2022-02-06:
I have already constructed the bellows from the beginning. The encoder holder too, thus the solder joints are relieved.
With me a decoder broke, when I was off the Knob and sat too tightly. 
[image]
[image]
[image]
In addition, I designed a Macro Keyboard and occupied with the buttons that are often used. I had the tried to publish, but probably did not work out.

FaqT0tum, 2022-02-06:
> I have already constructed the bellows from the beginning. The encoder holder too, thus the solder joints are relieved. With me a decoder broke, when I was off the Knob and sat too tightly. [image] [...]

I had also thought about drilling the upper part of the bellow, but so you can print it easily?

AFUDIrk, 2022-02-06:
I printed him from Filaflex 82a with Support.
But have used several assumptions until it worked.

helicopterguru, 2022-02-06:
> I have already constructed the bellows from the beginning. The encoder holder too, thus the solder joints are relieved. Then you can additionally attach him to the mother. Therefore, it is better to take a decoder with the longer axis and with thread. For me a decoder broke, when I have deducted the Knob and he sat too tightly. [image] [image] [image] In addition, I designed a Macro Keyboard and occupied with the buttons that are often used. I had the tried to publish, but probably did not work out.

Hey Buddy.

Ah very nice, thanks for coming in to reply. 
Have you posted the bellow STL somewhere I can download to give it a go?

I'm with you now on the pot support for the main shaft, makes snese to stop any possible movement  between the solder joints holding it to the PCB. WIll try and do a similare revision.

Good addition with the macro keys, I'd like to maybe add in two for copy & paste as my keyboard has these buttons and it's getting old now so needs replacing. Is that sketch around as well for me to take a look at?

Cheers.

Ian Contessa

FaqT0tum, 2022-02-06:
> I printed him from Filaflex 82a with Support. But have used several attempts until it worked.

you still could have curdled that part with a blade

AFUDIrk, 2022-02-06:
Here is my version to find. With all changes. I hope that's ok, I do not know how to make it accessible to you. 
 ](https://github.com/AFUDIrk/Orbion_Space_Mouse-2.2)

helicopterguru, 2022-02-06:
> Here is my version to find. With all changes. I hope that's ok, I do not know how to make it accessible to you. ](https://github.com/AFUDIrk/Orbion_Space_Mouse-2.2)

Awesome, thanks for sharing buddy.
When I can I'll look at the Arduino code to see what I need to import to add some macro buttons, as that could be handy from what I am seeing.

Ian Contessa

AFUDIrk, 2022-02-06:
Hello, yes I liked doing it.  I don't want to be without these macro keys
anymore.  They are very useful.

Mit freundlichen Grüßen
Dirk König

FaqT0tum, 2022-02-06:
If it is more convenient for you, I have created a discord channel, so we can share ideas better

FaqT0tum, 2022-02-06:
https://discord.gg/Kguv3ZSP

helicopterguru, 2022-02-06:
> Hello, yes I liked doing it. I don't want to be without these macro keys anymore. They are very useful. Mit freundlichen Grüßen Dirk König helicopterguru ***@***.***> schrieb am So., 6. Feb. 2022, 20:17:
> […](#)

For sure its going to be an addition ill add to my build some macro buttons. For now wrap it up as is and start using it.

Ian

helicopterguru, 2022-02-06:
> If it is more convenient for you, I have created a discord channel, so we can share ideas better

Hey buddy.

Cool, sounds like a good plan. I've joined the Discord. Will be good to see it expand and more development happen.

Ian

AFUDIrk, 2022-02-07:
Hi,
thank you.