sedmenti/Orbion_Space_Mouse_OLD #27: OLED stays dark

Davery92, 2022-02-05:
Good Morning, I've been following your project for a while and finally got all the parts necessary. I breadboarded everything this morning and uploaded the sketch after following your wiring schematic, but the OLED stays dark. I've tried two different screens and neither will turn on. The screen will randomly move due to having the joystick at an angle so I know something's wired right and working. Has this ever happened to you?

FaqT0tum, 2022-02-05:
> Good Morning, I've been following your project for a while and finally got all the parts necessary. I breadboarded everything this morning and uploaded the sketch after following your wiring [...]

the problem can be associated either with the wrong protocol, as there are identical OLEDs but with different controllers (SH1106 and SSD1306) or by the SDA and SCL cables reversed

Somebody1566, 2022-02-07:
Hey Davery, this is common with breadboards and oleds. It will most likely work if you solder it. Same issue happened to me.

Davery92, 2022-02-07:
You were absolutely right. I discovered this late last night while troubleshooting. I'll be soldering everything this week thanks for the help!

Somebody1566, 2022-02-07:
No worries! Happy to help

FaqT0tum, 2022-02-13:
> You were absolutely right. I discovered this late last night while troubleshooting. I'll be soldering everything this week thanks for the help!

If you need more help, I'll be happy to help you on Discord

closed 2022-02-13 by FaqT0tum