sedmenti/Orbion_Space_Mouse_OLD #32: 'Mouse' not found.

Cagedconner, 2022-02-14:
Arduino: 1.8.19 (Windows 10), Board: "Arduino Nano, ATmega328P"

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:8:0:

C:\Users\Conner\Documents\Arduino\libraries\Adafruit_SSD1306/Adafruit_SSD1306.h:63:0: warning: "BLACK" redefined

 #define BLACK SSD1306_BLACK     ///< Draw 'off' pixels

 

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:1:0:

C:\Users\Conner\Documents\Arduino\libraries\Adafruit_SH1106/Adafruit_SH1106.h:48:0: note: this is the location of the previous definition

 #define BLACK 0

 

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:8:0:

C:\Users\Conner\Documents\Arduino\libraries\Adafruit_SSD1306/Adafruit_SSD1306.h:64:0: warning: "WHITE" redefined

 #define WHITE SSD1306_WHITE     ///< Draw 'on' pixels

 

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:1:0:

C:\Users\Conner\Documents\Arduino\libraries\Adafruit_SH1106/Adafruit_SH1106.h:49:0: note: this is the location of the previous definition

 #define WHITE 1

 

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:8:0:

C:\Users\Conner\Documents\Arduino\libraries\Adafruit_SSD1306/Adafruit_SSD1306.h:65:0: warning: "INVERSE" redefined

 #define INVERSE SSD1306_INVERSE ///< Invert pixels

 

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:1:0:

C:\Users\Conner\Documents\Arduino\libraries\Adafruit_SH1106/Adafruit_SH1106.h:50:0: note: this is the location of the previous definition

 #define INVERSE 2

 

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:26:0:

C:\Program Files (x86)\Arduino\libraries\Mouse\src/Mouse.h:29:2: warning: #warning "Using legacy HID core (non pluggable)" [-Wcpp]

 #warning "Using legacy HID core (non pluggable)"

  ^~~~~~~

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:27:0:

C:\Program Files (x86)\Arduino\libraries\Keyboard\src/Keyboard.h:29:2: warning: #warning "Using legacy HID core (non pluggable)" [-Wcpp]

 #warning "Using legacy HID core (non pluggable)"

  ^~~~~~~

In file included from D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:30:0:

C:\Users\Conner\Documents\Arduino\libraries\Adafruit_SH1106/Adafruit_SH1106.h:48:0: warning: "BLACK" redefined

[... 440 lines cut ...]
                       ^~~~~~~~~~~~

D:\Downloads\Orbion_Space_Mouse-main (1)\Orbion_Space_Mouse-main\Orbion_Firmware\Orbion_FW_v0.26.2\Orbion_FW_v0.26.2.ino:689:23: note: suggested alternative: 'MOUSE_h'

           Mouse.press(MOUSE_MIDDLE);

                       ^~~~~~~~~~~~

                       MOUSE_h

Orbion_FW_v0.26.2:693:26: error: 'KEY_LEFT_GUI' was not declared in this scope

           Keyboard.press(KEY_LEFT_GUI);

                          ^~~~~~~~~~~~

exit status 1

'Mouse' not found. Does your sketch include the line '#include <Mouse.h>'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

 new arduino user here, likely user error but have tried a few things now and still get this error, even though that line is obviously present in the code

edifabio-72, 2022-02-14:
The microprocessor ATmega328P do not support mouse and keyboard library. It is essential to use ARDUINO PRO MICRO with ATMEL32U4 processor.

FaqT0tum, 2022-02-14:
have you read the guide carefully?  have you read all the disclaimes?  if you need more assistance go to Discord, you will find all the help you need https://discord.gg/KAbUJ2Gr

Cagedconner, 2022-02-14:
Thank you for the quick response, I must have received a different model arduino with my parts order. I assumed my "leonardo" was a pro micro from the store description

closed 2022-02-15 by Cagedconner