sedmenti/Orbion_Space_Mouse_OLD #45: Arduino Micro or Leonardo ?

fredmccr, 2022-02-26:
Hi,
I ordered and received my Arduino board from Aliexpress (same link as yours).
It's labeled "Pro Micro" on its top but when I get board info from Arduino IDE, it shows "Arduino Leonardo".
What kind of board should I select before flashing: "Arduino Micro" as stated in the readme file or "Arduino Leonardo" ?
Thanks

kc0rpm, 2022-02-26:
Mine showed up as a Leonardo also. Select micro when uploading code though.

fredmccr, 2022-02-26:
Thank you for your help.
I just flashed my board like you advised. Is it normal now that one of the two red LEDs remains always on ?

kc0rpm, 2022-02-26:
Yeah, thats normal.

Sisko4, 2022-03-04:
I think selecting it as Arduino Micro just briked it. It doesn´t finish uploading.

Sisko4, 2022-03-05:
Does it make any difference anyway if you choose Micro or Leonardo?
Every seller writes you need to choose Leonardo in the Board Selection.

kc0rpm, 2022-03-05:
I don't see why it would. Both boards use the same ATmega32U4 processor. The micro is just a miniaturized version. Of course I could be wrong.

FaqT0tum, 2022-03-08:
the arduino pro micro is a miniaturized leonardo

closed 2022-03-08 by FaqT0tum