sedmenti/Orbion_Space_Mouse_OLD #48: Fusion 360 files missing history?

nbering, 2022-03-07:
I would like to make some small corrections to the design to compensate for variability in parts. In this case, I have the following issues that I would have liked to correct for my build:

- The shaft on my encoder is much shorter than the design, so the hole in the Orbion knob barely grabs the top of the shaft, and it comes off easily.
- The alignment for the drill holes on my encoder is out-of-tolerance so that the knob bell doesn't match up to the edge of the base (the base sticks out to the side).
- The PCB for my joystick module is _slightly_ wider than the tolerance allowed in the design, so it doesn't seat into the Arduino's holder.
- I would have liked to adjust the holes for the heat-set insert to match some I keep in stock.

The first two issues could be addressed in the design itself if the base was inset into the knob with a wide vertical and horizontal tolerance. I don't expect to make that change myself.

I wanted to tweak the .f3d file to customize for my build, but everything was solid bodies. There doesn't seem to be a design timeline or any parameters I could adjust. Are they just missing from the export? I've never seen someone build something so complex in mesh mode, so can I assume there were sketches and a design history somewhere?

FaqT0tum, 2022-03-08:
history and status cleared before export due to file weight problems. I work without history, you can too. For the shorter shaft I suggest you buy a longer one because even if you solve the result it will always be disappointing.

nbering, 2022-03-09:
As was pointed out in #20, I bought the encoder linked in the bill-of-materials. I'd have been happy to buy a different part, but I didn't even know I needed to.

closed 2022-03-09 by nbering

FaqT0tum, 2022-03-09:
it is not your fault, but of the shortage that apparently allows the Chinese to pack whatever they find available in good warehouses in order to sell

I am trying to fix this situation but it is not easy to understand when the Chinese pack what you buy.  with other users we have already tried to stem this problem, the only solution is to extrude the 3 / 4mm internal slot of the D-shaft.  but unfortunately this way you have little support surface.  if you can wait for the new version I'm preparing

FaqT0tum, 2022-03-09:
[image]
[image]

FaqT0tum, 2022-03-09:
this is the new upgrade i am preparing.  will change before release but 90% will.  *so don't ask me for files in advance or bom